jps2000/BNO080 #5: BNO080 on Adafruit Feather M0

MadMadHoosier, 2018-09-12:
I'm an amateur programmer that just happened to be looking for BNO080 code as you posted the updated Arduino_BNO080_2.ino.  I like the ideas of the buttons for calibration and tare, but am obviously overlooking what ever is necessary to get this code to run on my Adafruit Feather M0.  Any suggestions?

PS:  The Example1-RotationVectors sketch runs fine on my M0, but it utilized the "SparkFun_BNO080_Arduino_Library.h".  Anyway, like I said: "amateur programmer"!

Sincerely, MadMadHoosier, Steve.

jps2000, 2018-09-19:
Hi MMH
I consider myself also a kind of amateur programmer. - don´t worry.
Unfortunately I do not have this board to test the code. Can you describe what it does / says when you upload the code?

MadMadHoosier, 2018-09-19:
Hi jps2000:
  Thank you so much for the reply.  It errors while compiling with an extensive list of error messages.  I'm guessing its related to the I2C library, but am not experienced enough to figure out why.  Hopefully something simple jumps out with the (long) error list below.  Thanks, Steve........
`
Arduino: 1.8.7 (Windows Store 1.8.15.0) (Windows 10), Board: "Adafruit Feather M0"

In file included from C:\Users\steve\Documents\Arduino\JPS2000bno080code\JPS2000bno080code.ino:49:0:

C:\Users\steve\Documents\Arduino\libraries\I2C/I2C.h: In function 'void scan_I2C()':

C:\Users\steve\Documents\Arduino\libraries\I2C/I2C.h:109:13: note: candidate 1: uint8_t I2C::read(int, int)

     uint8_t read(int, int);

             ^

C:\Users\steve\Documents\Arduino\libraries\I2C/I2C.h:108:13: note: candidate 2: uint8_t I2C::read(uint8_t, uint8_t)

     uint8_t read(uint8_t, uint8_t);

             ^

In file included from C:\Users\steve\Documents\ArduinoData\packages\adafruit\hardware\samd\1.2.2\cores\arduino/Arduino.h:36:0,

                 from C:\Users\steve\Documents\Arduino\libraries\I2C\I2C.cpp:57:

C:\Users\steve\Documents\Arduino\libraries\I2C\I2C.cpp: In member function 'void I2C::begin()':

C:\Users\steve\Documents\Arduino\libraries\I2C\I2C.cpp:91:9: error: 'PORTD' was not declared in this scope

     sbi(PORTD, 0);

         ^

C:\Users\steve\Documents\ArduinoData\packages\adafruit\hardware\samd\1.2.2\cores\arduino/avr/pgmspace.h:36:23: note: in definition of macro '_SFR_BYTE'

 #define _SFR_BYTE(n) (n)

                       ^

C:\Users\steve\Documents\Arduino\libraries\I2C\I2C.cpp:91:5: note: in expansion of macro 'sbi'

     sbi(PORTD, 0);

     ^

In file included from C:\Users\steve\Documents\Arduino\libraries\I2C\I2C.cpp:63:0:

C:\Users\steve\Documents\Arduino\libraries\I2C\I2C.h:83:51: error: '_BV' was not declared in this scope

 #define sbi(sfr, bit)   (_SFR_BYTE(sfr) |= _BV(bit))

                                                   ^

C:\Users\steve\Documents\Arduino\libraries\I2C\I2C.cpp:91:5: note: in expansion of macro 'sbi'

     sbi(PORTD, 0);

     ^

In file included from C:\Users\steve\Documents\ArduinoData\packages\adafruit\hardware\samd\1.2.2\cores\arduino/Arduino.h:36:0,

                 from C:\Users\steve\Documents\Arduino\libraries\I2C\I2C.cpp:57:

C:\Users\steve\Documents\Arduino\libraries\I2C\I2C.cpp:95:7: error: 'TWSR' was not declared in this scope

   cbi(TWSR, TWPS0);

       ^

C:\Users\steve\Documents\ArduinoData\packages\adafruit\hardware\samd\1.2.2\cores\arduino/avr/pgmspace.h:36:23: note: in definition of macro '_SFR_BYTE'
[... 473 lines cut ...]
C:\Users\steve\Documents\Arduino\libraries\I2C\I2C.cpp:705:14: error: 'TWEN' was not declared in this scope

   TWCR = _BV(TWEN) | _BV(TWEA); //reinitialize TWI 

              ^

C:\Users\steve\Documents\Arduino\libraries\I2C\I2C.cpp:705:18: error: '_BV' was not declared in this scope

   TWCR = _BV(TWEN) | _BV(TWEA); //reinitialize TWI 

                  ^

C:\Users\steve\Documents\Arduino\libraries\I2C\I2C.cpp:705:26: error: 'TWEA' was not declared in this scope

   TWCR = _BV(TWEN) | _BV(TWEA); //reinitialize TWI 

                          ^

exit status 1
Error compiling for board Adafruit Feather M0.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

jps2000, 2018-09-19:
yeah its all about I2C
You can only use the Arduino_BNO080_2.ino file. It uses the wire.h lib for the I2C bus. Both others use the I2C lib which will not work on a M0

MadMadHoosier, 2018-09-19:
Wow, not only am I amateur, but not observant enough to notice you had a wire.h version :-(  It is now loaded and spitting data to the monitor (after I moved the SDA & SLC to A4 & A5) and I'll get back to playing with it next week.  

Probably a rookie question I can figure out when I return next week, but this is the first I2C sketch I've seen use analog pins instead of the pins labeled SDA & SLC.  I'm not familiar with the Nano, so I 'm guessing it does not have the separate pins - or is it something else?

Thanks so much for your patients and feedback!

closed 2018-10-24 by jps2000