jps2000/BNO080 #6: UART communication

m2oswald, 2018-11-30:
Hi there.  I am using the Hillcrest FSM300 board with an Arduino Mega and your code works perfectly.  I was wondering if you have ever tried to communicate via UART instead of I2C?  I tried converting the BNO080_2 code to talk to the FSM300 through the Mega's Serial1 connection, but have had no luck at all. Serial1.available never goes > 0, so I don't know whether I'm not sending the setup commands correctly or not listening correctly.  I tried looking through Hillcrests' Nucleo code but that hasn't helped at all, and their tech support hasn't either.

Any help/advice you could give would be greatly appreciated, as I've been banging my head against the wall for a few days now and have gotten nowhere (except a headache).

Michael

jps2000, 2018-11-30:
Hi Michael
I never tried that
In Hillcrest data sheet there is  a description of how to use the uart interface. it is meant for robot vacuum cleaners. Would try that to see if interface works. You need a 3v Uart  of course.
This is what the spec says
1.3.3.1 UART operation
The UART is configured for 3Mkb/s, 8 data bits, 1 stop bit and no parity. The UART protocol relies on an idle line being ‘high’. A transmission is started with the assertion of a start bit (pulling the line low), followed by the data, LSB first. After the data segment is sent (in this case 8-bits), the line is pulled high (the stop signal) for a minimum number of bits (1 for the BNO08X) to indicate end of that segment. Bytes sent from the host to the BNO08X must be separated by at least 100us. Bytes sent from the BNO to the host have no extra spacing

I never heard about this 3Mkb/s?
sanity check TX goes to RX and vice versa?

m2oswald, 2018-11-30:
Thanks so much, I appreciate the quick reply.  I actually just tried the
UART-RVC mode, and that works.  So that means that I'm able to read the
comm port ok, my problem is that I'm not writing to it correctly (or, at
least, not in a way that the BNO080 understands).  I'm wrapping my SHTP
messages in the UART 'framing bytes' as mentioned at the very end of
1.3.3.1 and detailed in Hillcrest's 1000-3535 document.  I'm running at
3000000 baud (which seems way too high, but is specified everywhere so I'm
just going with it).  And I even tried swapping the RX and TX, just in
case.  Nothing.

As far as the serial comm details (pulling the line high/low and separating
by a 100us pause) I'm making the assumption that the Arduino's Serial
library is doing these things correctly.  And I guess that since I'm able
to read data in the UART-RVC mode, maybe that's true.

Thanks again for your help.  It half works now.  I'll go back to banging my
head against the wall for awhile.

Michael

jps2000, 2018-11-30:
not sure but can the  mega do this speed?
After power on/ reset better wait for 600ms or longer 
Have you digested the SHTP transport protocol chapter 4?

ramiss, 2018-11-30:
I commend you for trying this. Unfortunately others have confirmed the maga only handles at most 250,000 bps.

Are you just trying for science or do you really not have i2c available?

m2oswald, 2018-11-30:
Of course, I was so focused on finding the problem with the FSM300 that I
didn't even think about limitations on the Mega.  Ok, well that's my answer
then.  And it explains why UART-RVC works - it communicates at 115200 baud.

Thank you for the help and advice, I really appreciate it.

Michael

m2oswald, 2018-11-30:
Sorry, I forgot to respond to ramiss.  For my application, the arduino and the sensor will be separated by 2-4 m, which I think is pushing it for reliable I2C.  And this is going to be mounted on platform in a fairly remote location, so once it's in place it really needs to work because I can't go back in to change it.

ramiss, 2018-12-01:
I agree with Peter. I2C will be ok at 4M with shielded cables and pullups. You may have to reduce transmission rate, but on the whole it would be no less reliable than using serial. The I2C repeater should guarantee the distance.

m2oswald, 2018-12-03:
My equipment is actually going to be underwater, off the coast of northern Scotland.  So I'm limited in the cables that I can use as they will be exposed to seawater.  The arduino is housed in a watertight enclosure, and the FSM300 will be potted and mounted on a adjustable platform a meter or two away.  I did the calculations based on the capacitance of the cable I plan to use, and I think 2 m would be ok with small (1.2k) pull-up resistors on the SDA and SCL lines.  But it just seems like there's more places something could go wrong with I2C compared to serial.
I spoke with Hillcrest tech support and the FSM300 is stuck at 3Mbps for regular UART mode, so that's out.  But I think the UART-RVC mode will be good enough for my purposes, and that only requires 115200 baud.
Thanks again to both of you for all of your help and advice, I appreciate it.

jps2000, 2018-12-03:
Welcome.
At any moment you can try the cable you intended to use with I2C bus and see what happen.
With UART RVC  the BNO080 is just spiiting data. So you can not  do tare  etc.

closed 2018-12-04 by jps2000

choprahetarth, 2019-10-09:
Hi
I was trying to work my BNO080 (SparkFun breakout board), so I was wondering if there's some source code on how to interpret the data received from the IMU?
Thanks

jps2000, 2019-10-10:
Hi I have not seen so far a software stack using BNO080 with UART. Not sure what you meant with "interpret the data".

choprahetarth, 2019-10-16:
Interpret as in changing the packet to human-readable format

choprahetarth, 2019-10-16:
Anyways, I tried making one code to do so... not perfect but would want suggestions for it - https://github.com/choprahetarth/Sparkfun_VR_IMU_UART_ROS/blob/master/serial1.py

jps2000, 2019-10-16:
unfortunately I do not talk python. Just C++ hobbyist level

choprahetarth, 2019-10-16:
that's okay... let's hope someone comes along and makes a c++ version of the same

jps2000, 2019-10-17:
My suggestion for max functionality would be to output quaternions and linear acceleration 
From the quats you have orientation in space (Eulers) and gravity vector. Linac indicates a movement (integral is velocity)
Additionnally cal status info is helpful

jps2000, 2019-10-30:
No there is a very effective filter. Such a behaviour is known when calibration is poor.
Best practice is
# Use rotation vector 0x05
# 200-400Hz update rate
# Degauss the board.
# Be careful with indoor use
# Leave calibration setting as is
# In setup call save_periodic_DCD();
# calibrate the board including  iron parts in the vicinity

Saicharan036, 2021-07-17:
Hello,
 I am working on FSM300 Interfacing with Raspberry pi compute module 4 and I am getting the similar kind of issue i am able to read the data in UART-RVC mode and unable to read the data in UART-SHTP Mode. I have connected the hardware as 
 Fsm300      Raspberrypi
 Vdd--------->Vdd
 Gnd--------->Gnd
 Txd --------->Rxd
 Rxd--------->Txd
 PS1--------->Vdd
 PS0--------->Gnd
 GPIO11----> BootN
 GPIO13----->Nrst

 
 I have Installed the libraries adafruit_circuitpython_BNO08x and run the basic code from the Examples "https://github.com/adafruit/Adafruit_CircuitPython_BNO08x/blob/main/examples/bno08x_simpletest_uart.py " Is there any Initial setting need to be done before Reading the data .?

Any advice you can give would be appreciated

Saicharan

jps2000, 2021-07-17:
I will answer on adafruit forum

Saicharan036, 2021-07-17:
Hello,
  The below is the link of the issue in Adafruit forum
  https://forums.adafruit.com/viewtopic.php?f=19&t=181183
  
saicharan