jps2000/BNO080 #8: Thank you

fred9999, 2019-09-13:
This obviously is not an issue, but I did not know where else I could express my gratitude. 

I just want to say very big thank you jps2000 for sharing the code and being so responsive. I went through earlier issues and your comments and I was able to solve my issues with BNO080 drift at last. The trick that solved the problem was degaussing it and calibrating at site. Thank you for the advice!

Keep up the good work,

Fred

ramiss, 2019-09-13:
Could you expand on your method for degaussing?  Did you literally just orbit a magnet around it?

fred9999, 2019-09-13:
I just happened to have a demagnetizer tool, working from mains AC. Bought from usual Chinese suppliers. I switched the demagnetizer on at the distance of 1m, slowly moved it to the IMU, held it on the IMU for a couple of seconds and slowly moved away and switched it off at the distance of 1m.

You can easily make the demagnetizer yourself. Mine is just a mains power transformer without a cross bar.

Hope this helps,
Fred

jps2000, 2019-09-13:
You´re welcome
To explain degaussing:
Hard iron can be magnetized when it comes into contact with a magnet or even just when exposed to a strong magnetic  field. 
Here
https://en.wikipedia.org/wiki/Magnetic_hysteresis
you see what happens when a ferromagnetic material is exposed to a magnetic field.
Look at the graph with the loop. The loop does not fall back to zero once the magnetic field is gone.
There is remanence magnetism in one or the other direction. That principle is the base for magnetic tapes and disks for data storage.
Now the idea of degaussing is to get somehow in the origin of the curve.
This can be achieved by running multiple times through that loop with diminishing magnetic field.
It is done by an electromagnet powered with alternating current (mains)
You can use for that the coil of an magnetic valve operating with mains. You need to replace the valve by an iron screw with. DO NOT OPERATE THE COIL WITHOUT SCREW ! It will get hot and melt down.
Then switch the coil on in a distance of about 1m and bring it close to the parts to be degaussed. Then remove the coil again about 1m  and switch it off.
You can try this by magnetizing a screw driver with a strong magnet, check for magnetism and then degauss it as described and check again.
Degaussing was also used in CRT screens with a large coil around the mask. The was a PTC  resistor in series. When  the TV was switched on the PTC was cold and hence has low resistance. High current passes the coil and the PCT heats up so current went down rapidly to very small values.
A faulty degaussing caused strong color  distortions as the electron beam was deflected by the magnetic field.
Standard SMD components  have ferromagnetic contacts or housing (xtal). When they are close to the IMU chip ( as they should)  they can introduce hard iron offset.  There are special SMD for MRI applications available but  they are more expensive

https://en.wikipedia.org/wiki/Degaussing

jps2000, 2019-09-13:
Great 
just read your comment after writing this

ramiss, 2019-09-13:
Thank you both. Great info!   
It does been the question however... If the IMU is supposed to do a magnetic calibration, then is degaussing really necessary?

jps2000, 2019-09-13:
The problem is that yo do not know if it is necessary. You kan have a look with a magnetic compass 2-3 meter away  if it points in the same direction. However some degrees of deviation you can hardly see.
Degaussing just clears a potential problem. It is like sterilizing surfaces. There may or may not dangerous bacteria.

ramiss, 2019-09-13:
That makes sense, my application is compounded by 4 motors within inches of the IMU,  So,  I'm fascinated to know if the magnetic sterilization would slowly change on the IMU whether I degaussed or not.  To expand your analogy,  its like sterilizing a surface that constantly has sick people around it.... so is it worth the effort if the surface can account for the bacteria?

Fun thought exercise regardless.

fred9999, 2019-09-13:
With motors so close to IMU magnetometer data will be disturbed greatly and it will be totally unreliable. In my case I had wires with 30A current near the IMU and I had to move the IMU 70cm away from them to get clean magnetometer data.

jps2000, 2019-09-13:
Usually drones use 6dof IMUs, Baro for height  and GPS for compass so no magnetometer at all
High DC currents are a problem. Motors may be not so  influencing because they operate at high frequencies. The magnetometer accept just 10 to 20  Hz or so. So they are not influenced by mains fields or switchmode power supplies. 
Motors also have magnets in and have an iron core that may be a problem.

fred9999, 2019-09-13:
By the way Peter as you mentioned getting positioning data without magnetometer I have a question to you.

I am using the BNO080 in a boat to keep it on course for not long periods - max 30min. Initially I was very exited about the GameRotationVector because it does not use the magnetometer data at all and I could place the IMU where it was the most convenient but as said it was near the wires with high DC current. Bench tests with GameRotationVector were very promising - drift was only a couple of degrees for 30min period. But on the water it started to drift 15-30 deg/min after about an hour. Resetting it by switching off and on did not help. Then I cancelled automatic calibration in the code but that did not help either. It had to lie still for at least 5 minutes to work properly again. But on the water with wind and waves that is obviously impossible.

So Peter, is there a way to use the GameRotationVector for about 30 minutes and then reset the BNO080 into initial state with initial calibration so it could work properly for next 30min period.

Thank you beforehand.

jps2000, 2019-09-13:
When you want to keep a boat on a course then you need heading ( yaw).  The game rotation vector has 6dof so you need a start direction  . Whenever you  reset the device it starts with quaternions 1,0,0,0 in this position. 
Switching off autocal can be deterious I found . So may be better not touch that.
You need to watch the cal status that should not go below 2. When the sensor is moved to long and too much this can be a problem. 
The 5 min wait seems quite long to me.  temperature change may be a  problem. I considered to mount the bno in a thermostat or to heat it with a resistor to 50 deg (above ambient) and to measure internal temperature and keep it rather constant. but I never tried this.
But you may do something with your 30A wires. 
You may twist the wires.  This lead to small loops with changed orientation that compensates magnetic field roughly . 
A better solution would be to make a self made coaxial cable. Put  one isolated wire  through  a copper tube and let the forward current  flow on the inner wire  and the backward current outside in the copper tube. Coaxial arrangements fully compensate magnetic fields.
You need to do this only in close vicinity (1m) to the BNO.

fred9999, 2019-09-14:
Thank you very much for the quick reply. Your suggestions definitely make sense.

I suspect that it starts to drift because it can not auto calibrate itself (happens in every 5 minute) because it is in constant motion. I looked from manual for the complete reset and found this:

6.4.10Clear DCD and Reset(0x0B)
The recommended sequence for completely resetting DCD state is:
1.Reset hub (via pin toggle, reset command, or Clear DCD and Reset Command)
2.Delete flash copy of DCD via FRS: see 6.3.3.
3.Issue Clear DCD and Reset Command.

I never figured out how to do this "Delete flash copy of DCD via FRS" part. 

Maybe you happen to know how to do that? Thanks.

jps2000, 2019-09-14:
I do not think reset DCD is helpful.
My current best practice is the following: ( Rotation vector, 9DOF)
In setup call 
save_periodic_DCD();              // saves DCD every 5 minutes ( only if cal = 3) 
( do not call ME_cal(1,1,1,0); ) 
In loop running  after about 1 minute  you can do calibration  movements.
For MAG it seems beneficial to do slow rotation  in both directions (2sec/ turn)
Do this until stat is 3 is achieved
Then press a button that  does
save_DCD();                                     // store cal in flash
leave ME_cal unchanged.

Now every 5 min there is an auto save DCD done ( only if  stat =3) so it continuously adapt to drifts etc.
After every reset the actual DCD is re-loaded
You find all these functions in 
Arduino_BNO080_2.ino

fred9999, 2019-09-15:
Thank you very much for your reply and suggestions. I will try it next time I'll get onto water.

Just FYI here are the settings it is working at the moment with stable results without drift. A couple of days ago I was on water 12 hours and at the end of that period it was still working strong.

-degauss and calibration on boat
-automatic calibration cancelled for all sensors
-RotationVector
-BNO080 connected to Arduino via cable 1.5m
-pullup resistors on SDA and SCL lines 2.2k
-Wire.h library
-I2C data rate 100kHz

closed 2019-10-13 by jps2000