jps2000/BNO080 #10: Yaw, pitch and roll - BNO080

GianmarcoMicrotech, 2019-12-10:
Hi,
I have a new problem, with this buffer 'cargo' i obtain this yaw,pitch and roll value:
0x17 0x80 0x03 0x9B 0xFB 0x17 0x00 0x00 0x00 0x05 0xB2 0x03 0x00 0xA8 0x16 0x21 0xF1 0xF6 0xDF 0x53 0x30 0xCF 0x01

a=0.755065
b=0.354003
c=-0.23236
d=-0.50061
argument=-0.00354
yaw=336.9
pitch=-0.2
roll=50.1

But the pitch value can't be about equal to 0, why the sensor is positioned like this:

                      \
                        \  -> BNO080
                          \
             ---------------------------- -> DESK

The pitch should take the value about 70-80. I posted my code:

/* VARIABLES */

float 		Q0 , Q1 , Q2 , Q3 ;

float 		H_est ;	// heading accurracy estimation
uint8_t 	stat_ ;	// Status (0-3)

int 			len = 0 ;
uint8_t 	next_data_seqNum = 0x0A ;	// reading next data sequence number start with 10 to catch up earlier
uint8_t 	seqNum_2 = 0 ;	// writing sequence number channel 2 
uint8_t		cargo [ 300 ] = "" ;                                

const	uint8_t		quat_report = 0x05 ;	// defines kind of rotation vector (0x05), geomagnetic (0x09), AR/VR (0x28),
																			// without magnetometer : game rotation vector (0x08), AR/VR Game (0x29)
																
#define reporting_frequency       100	// reporting frequency in Hz
// report interval need to be longer than loop duration not to miss any report ! --> code checks for incrementing seqNum  
// note that serial output strongly reduces data rate

uint8_t B0_rate = 1000000 / reporting_frequency ;	// Calculate LSB (byte 0)
uint8_t B1_rate = ( 1000000 / reporting_frequency ) >> 8 ;		// Calculate byte 1

void 		get_QUAT ( void )
{                                                 	
	memset ( cargo , 0 , sizeof ( cargo ) ) ;
	
	nrf_drv_twi_rx ( &m_twi , BNO_ADDRESS , cargo , 4 ) ;	
  
	// Verifico se ho ricevuto il messaggio
  if ( check_cargo_buffer_isEmpty ( cargo , 4 ) == false )
	{     
		memset ( cargo , 0 , sizeof ( cargo ) ) ;
		nrf_drv_twi_rx ( &m_twi , BNO_ADDRESS , cargo , 23 ) ;	
		
    // to reduce I2C bus and computing --> read header first		      
		if ( ( cargo [ 9 ] == quat_report ) /*&& ( ( cargo [ 10 ] ) == next_data_seqNum )*/ )
		{      
			int16_t 	q0 , q1 , q2 , q3 , h_est ; 			
			float 		a , b , c , d , norm ;
			 
			// Check for report and incrementing data seqNum
			next_data_seqNum = ++ cargo [ 10 ] ;	// Predict next data seqNum    
			stat_ = cargo [ 11 ] & 0x03 ;		// bits 1:0 contain the status  
	
			q1 = ( ( ( int16_t ) cargo [ 14 ] << 8 ) | cargo [ 13 ] ) ; 
			q2 = ( ( ( int16_t ) cargo [ 16 ] << 8 ) | cargo [ 15 ] ) ;
			q3 = ( ( ( int16_t ) cargo [ 18 ] << 8 ) | cargo [ 17 ] ) ;
			q0 = ( ( ( int16_t ) cargo [ 20 ] << 8 ) | cargo [ 19 ] ) ; 

			// Patch for rarely occurring wrong data from BNO of unknown reasons. --> Check if q(0,1,2,3) is not unity vector
   
			// Apply Q point (quats are already unity vector)
			a = q0 * QP(14) ; b = q1 * QP(14) ; c = q2 * QP(14) ; d = q3 * QP(14) ;	
			norm = sqrtf ( a * a + b * b + c * c + d * d ) ;
			
			// Skip faulty quats; margin is empirically determined
			if ( abs ( norm - 1.0f ) > 0.0015 ) return ;	
    
			// Update quaternions
			Q0 = a ; Q1 = b; Q2 = c; Q3 = d ;	
			
			// bits 1:0 contain the status (0,1,2,3)
			stat_ = cargo [ 11 ] & 0x03 ;		 
 
			// Heading accurracy only in some reports available
			if ( ( quat_report == 0x05 ) || ( quat_report == 0x09 ) || ( quat_report == 0x28 ) )	
      {  				
				// Heading accurracy estimation 
				h_est = ( ( ( int16_t ) cargo [ 22 ] << 8 ) | cargo [ 21 ] ) ;	 
				
				// Apply Q point 
				H_est = h_est * QP(12) ;	
				
				// Convert to degrees 
				H_est *= radtodeg ;	
	
      }
    
			// Calculate euler angles                
			yaw   =  atan2f ( Q1 * Q2 + Q0 * Q3, Q0 * Q0 + Q1 * Q1 - 0.5f ) ;
			
			// May be > 1 due to rounding errors --> undefined
			float	argument = 2.0f * ( Q1 * Q3 - Q0 * Q2 ) ;	
			
			if ( abs ( argument ) > 1.0f ) 
				argument = 1.0f ; 
			
			pitch =  asinf ( argument ) ; 
			
			roll  =  atan2f ( Q0 * Q1 + Q2 * Q3, Q0 * Q0 + Q3 * Q3 - 0.5f ) ;

			// Correction of the y axis direction
			yaw += M_PI * 0.5f ; 
			
			if ( yaw < 0 ) yaw += 2.0f * M_PI ;
			if ( yaw > 2.0f * M_PI ) yaw -= 2.0f * M_PI ;  

			yaw   *= radtodeg ;
			yaw    = 360.0f - yaw ;
			pitch *= radtodeg ;
			roll  *= radtodeg ;
			
			// Ci sono nuovi dati, li invio via BLE
			update_data_BNO080 = true ;

		}
				
	}
	
}

Where is the problem ?

Regards

jps2000, 2019-12-10:
Hi
when you put your numbers for a,b,c,d in the formula for argument you do get a very different value?
do you have  not enough free  memory ?

GianmarcoMicrotech, 2019-12-10:
I have many memory free. 

I get these values printing them with the debugging, while the sensor is working:
a=0.755065
b=0.354003
c=-0.23236
d=-0.50061
argument=-0.00354
yaw=336.9
pitch=-0.2
roll=50.1

These value ( a,b,c,d, argument, yaw, pitch and roll ) are obtained from this buffer:
0x17 0x80 0x03 0x9B 0xFB 0x17 0x00 0x00 0x00 0x05 0xB2 0x03 0x00 0xA8 0x16 0x21 0xF1 0xF6 0xDF 0x53 0x30 0xCF 0x01

Why the algorithm calculate a pitch value so low ?

Thanks you,

Regards

jps2000, 2019-12-10:
Ok
my post above is not correct because I missed one minus sign.
Only thing I  can argue is that the cal  is still poor despite it is showing 3.
What happens at smaller tilt angles? is then everything ok?  or  are the angles somewhere not correct.
Have you applied tare somewhere? this can modify your sensor even permanently.
so check first  if yaw is correct . than apply pitch +/-45 and then roll +/-45. is that ok?

GianmarcoMicrotech, 2019-12-10:
"my post above is not correct because I missed one minus sign." Where is the error in the code ?

"What happens at smaller tilt angles? is then everything ok? or are the angles somewhere not correct.
Have you applied tare somewhere? this can modify your sensor even permanently.
so check first if yaw is correct . than apply pitch +/-45 and then roll +/-45. is that ok?"  
Can you explain better at me ? 

Thanks you,

GianmarcoMicrotech, 2019-12-10:
Update information

If the sensor is resting with the board long side on the desk in a vertical position, i obtain these results:
q1=0x008F
q2=0x0A78
q3=0xFF54
q0=0x3F22

a= 0.98
b=0.008
c=0.163
d=-0.010

yaw=271
pitch=90 (value fixed, strange)
roll=0.8

If the sensor is positioned horizontally on the desk, i obtain these results:
q1=0xEC33
q2=0xD366
q3=0xF221
q0=0x2704

a=0.609
b=-0.309
c=-0.696
d=-0.216

yaw=158
pitch=79
roll=-155

If the sensor is resting with the board short side on the desk in a vertical position, i obtain these results:
q1=0xD5AD
q2=0x11C3
q3=0x0D4C
q0=0x2A92

a=0.665
b=-0.661
c=0.277
d=0.207

yaw=276
pitch=-40
roll=-92

These results are corrects?

Thanks you,
Regards

jps2000, 2019-12-11:
This clarifies your problems.
Having the sensor flat on the table there should be yaw xxx (depending on orientation) pitch~0, roll~0
if this is not the case you have done tare or tare persist when the sensor hat an arbitrary (undefined orientation.
Better do not use tare at all because then you change yaw pitch roll orientation relative to the BNO housing.
You may repair this by clearing the tare setting ( I do not know by heart at the moment how to do that --> look into data sheet)
or you can do tare by placing the sensor flat an in direction north on  a table after careful calibration. and then do tare persist **just once**.
tare does the following:
It sets quaternions to 1,0,0,0   at the actual orientation (any) of the sensor. 
tare now is not remembered after reset. tare persist is remembered ( as the name says)
So you have altered the sensor quaternion output  forever.

GianmarcoMicrotech, 2019-12-11:
Ok, therefore if i cancel the Tare with this function i resolve my problem ?
void 		clear_TARE ( void )	
{
	// set reorientation
  uint8_t tare_clear [ 16 ] = {16,0,2,seqNum_2,0xF2,0,0x03,0x02,0,0,0,0,0,0,0,0};                

	// register adress is the LSB of the header 

	nrf_drv_twi_tx ( &m_twi , BNO_ADDRESS , tare_clear , 16 , false ) ;	
	++seqNum_2; 

}

The results (yaw, pitch and roll) that i previously posted are correct ?

Thanks you,

jps2000, 2019-12-11:
Probably yes
if you have the BNO flat on the desk and when you get pitch , roll very close to zero  it is ok
I can not check your results easily but they should be correct.

GianmarcoMicrotech, 2019-12-11:
Ok, when the BNO is flat on the desk a sometimes the pitch alternates low values (3.7) a the value fix 90, why ? 90 is not correct. This problem occurs continuously.

Thanks you,

Regards

jps2000, 2019-12-11:
I do not understand.
Can you post a table showing  yaw, pitch, roll, hest and stat (for example every second) when the BNO is flat on the table

GianmarcoMicrotech, 2019-12-11:
ok, 
Results:
yaw=69.6
pitch=90
roll=-2.7

Time:2seconds
yaw=69.4
pitch=3.7
roll=-2.7

Time:3seconds
yaw=69.5
pitch=90
roll=-2.7

Time:3seconds
yaw=69.7
pitch=3.6
roll=-2.7

Time:4seconds
yaw=70.1
pitch=90
roll=-2.7

Time:5seconds
yaw=69.5
pitch=3.6
roll=-2.7

Time:6seconds
yaw=69.6
pitch=90
roll=-2.7

Thanks you,

Regards

jps2000, 2019-12-11:
that is more than strange. Can you stop bluetooth and look if it is the same?

GianmarcoMicrotech, 2019-12-11:
I have execute the test stopped the bluetooth, but the result is the same, the algorithm return wrongly value pitch equal to 90. Why the sensor return this  wrongly value ?

Thanks you,

Regards

jps2000, 2019-12-11:
I have never seen such results.
 May be you  have a problem with the I2C bus: It can be that you do not read correct data from the BNO although it  is sending correct data
Which controller do you use
which BNO board do you use
Which pullups do you have?
etc
First you need 400kHz. Then you are using  in your program  a combination of older and newer software now. I do not have this ++seqNum_2 etc .
Please copy carefully 
Arduino_BNO080_2.ino
or that 
nrf52 feather_BNO080.ino
example without anything else and see if it works

GianmarcoMicrotech, 2019-12-11:
OK, i have checked my function are equal to those implemented in the Arduino_BNO080_2.ino example. When the sensor return pitch value equal to 90, this is buffer received to sensor:

0x17 0x80 0x03 0x8F 0xFB 0xA1 0x00 0x00 0x00 0x05 0xE8 0x03 0x6D 0xB5 0xFD 0x10 0x01 0x59 0xC1 0xD0 0x0C 0xF4

jps2000, 2019-12-11:
What may happen is that the buffer is not fully rewritten.
This you can test  by clearing the buffer ( writing zeroes):
void get_QUAT()
{
int16_t q0,q1,q2,q3,h_est;                        // quaternions q0 = qw 1 = i; 2 = j; 3 = k;
float a,b,c,d,norm;
for( int i = 0; i<24; i++)  cargo[i] =0;         // clear cargo just to make sure nothing old is remaining
....
With that try again

jps2000, 2019-12-11:
IN your example from above you have written:
<<<<<<
When the sensor return this wrong value, the algorithm calculate these values:
q0=0x0AB8
q1=0xFDA7
q2=0x011E
q3=0xC0F5

a=0.167
b=-0.0366
c=0.017
d=-0.0985

norm=1.00000823
stat_=0x03
H_est=4.825

yaw=70.611
pitch=90
roll=-2.681

>>>>>
you see that norm is not correctly calculated
norm = sqrtf ( a * a + b * b + c * c + d * d ) ;
		
		// Skip faulty quats; margin is empirically determined
		if ( abs ( norm - 1.0f ) > 0.0015 ) return ;

GianmarcoMicrotech, 2019-12-11:
In my results a,b,c,d are corrects ?
It's not correct only the calculation of norm value ?

GianmarcoMicrotech, 2019-12-11:
Update :
I have commented these two code line and the pitch value remain constant, the pitch value does no longer assumes the value equal to 90, why ?
if ( abs ( argument ) > 1.0f ) 
     argument = 1.0f ;  

Thanks you,
Regards

jps2000, 2019-12-12:
Please post your current setup and your current  getquat subroutine
or - better- your complete code
and please answer  the questions 
Which controller do you use?
which BNO board do you use
Which I2C pullups do you have?

GianmarcoMicrotech, 2019-12-12:
Which controller do you use? I use the board nrf52840 from the Nordic.
which BNO board do you use? The name is FSM30X.
Which I2C pullups do you have? Yes, in the board there is a pullups, but i don't knows give you another informations because i aren't electrical engineer.I'm sorry.

My code :
#include <stdint.h>
#include <string.h>
#include <stdbool.h> // add
#include <stddef.h>	// add
#include <stdio.h>
#include <math.h>

#include "BNO080.h"

/* DEFINE */
#define BNO_ADDRESS             0x4A		// Device address when SA0 Pin 17 = GND; 0x4B SA0 Pin 17 = VDD

#define REGISTER_ADDRESS				21			// 21 =  predicted lenght incl header 
 
#define QP(n)                       (1.0f / (1 << n))          // 1 << n ==  2^-n
#define radtodeg                    (180.0f / M_PI)

/* VARIABLES */

float 		Q0 , Q1 , Q2 , Q3 ;

float 		H_est ;	// heading accurracy estimation
uint8_t 	stat_ ;	// Status (0-3)

int 			len = 0 ;
uint8_t 	next_data_seqNum = 0x0A ;	// reading next data sequence number start with 10 to catch up earlier
uint8_t 	seqNum_2 = 0 ;	// writing sequence number channel 2 
uint8_t		cargo [ 300 ] = "" ;                                

const	uint8_t		quat_report = 0x05 ;	// defines kind of rotation vector (0x05), geomagnetic (0x09), AR/VR (0x28),
																			// without magnetometer : game rotation vector (0x08), AR/VR Game (0x29)
																
#define reporting_frequency       400	// reporting frequency in Hz
// report interval need to be longer than loop duration not to miss any report ! --> code checks for incrementing seqNum  
// note that serial output strongly reduces data rate

/* EXTERN */
extern const	nrf_drv_twi_t m_twi ;
extern float	yaw ;
extern float	pitch ;
extern float	roll ;
extern bool		update_data_BNO080 ;

// Nota : funzioni riprese dal file 'Arduino_BNO080.ino' e da 'nrf52 feather_BNO080.ino'

 /**************************************************************************/
/*!
    Inizilizzo il sensore
*/
/**************************************************************************/
bool	BNO080_init ( void )
{
	uint8_t	tmp_buff [ 25 ] = "" ;
	
	nrf_delay_ms ( 1000 ) ;
	
	// Resetto il BNO080
	
	reset_BNO ( ) ;
	
	// Pausa di attesa
	nrf_delay_ms ( 1000 ) ;
	
	// Inizalizzo l'IMU
	initialize ( ) ;
[... 434 lines cut ...]
		/* INITIALIZE THE INU */
		error_system = BNO080_init ( ) ;
		
    // Verifico se devo comunicare lo stato di errore.
		// Nota: Controllo solo error_system in questo caso.
		blink_error ( error_system , NRF_SUCCESS ) ;	

     // Start execution.
    application_timers_start ( ) ;
    advertising_start ( erase_bonds ) ;
			
		// Cancello la Tara
		clear_TARE ( ) ;
		
    // Enter main loop.
    for (;;)
    {			
				/******************* IMU. ***************************** 
				*/							
				// IMU BNO080
				get_QUAT ( ) ;
	
    }
		
}

jps2000, 2019-12-12:
I am sorry this is a combination of many  programs. I can not  and do not want to dig into it.

closed 2019-12-17 by jps2000