jps2000/BNO080 #12: Question (actually looking for your opinion) re: reliability

ramiss, 2020-06-10:
Hi There jps2000!

You probably remember me and I wanted to say that I have the BNO080 running fine for my testing thus far. I just so happened to come back here and re-read your repository readme and it got me thinking...  Is it fair to say that all of the risk you mention with this sensor (ie using indoors, degauss practices etc) are mostly directed at the heading / magnetometer sensors?  

I ask because I am using the BNO080 primarily for a balancing robot, which relies on a stable X,Y (accel and gyro - but I think it also uses the magnetometer). The heading is secondary and not relied on in my use case.   In your opinion is the X,Y output on this sensor at risk because of the advanced housekeeping required?  If so, maybe I should downgrade to something more reliable such as a MPU-6050?   Or....once calibrated, are you fairly certain that the BNO080 X,Y would at least remain stable even if my sensor suddenly moved next to a microwave oven :) 

I love the BNO080 for the precision and speed, but I worry that it may be overkill and have reliability issues in the real world. 

Your opinions (and I know they are just that) are appreciated.

Thanks!

Richard

jps2000, 2020-06-10:
Hi Richard,
I have not worked with balancing robots. Probably they need just a fast gyro (and accelerometer) To get reliable  quaternions for orientation in  you need a good fusion sw and there the BNO is suoerior. You can at any time use the BNO in 6dof mode in case magnetometer is not reliable enough in your environment.
You may consider a fast gyro  like the MPU dedicated for balancing the robot and a BNO for orientation. Then you keep tasks separated.
Just little more power consumption but negligible to motors I think.
I do not think that a sudden  appearing piece of iron will compromise the gravity vector significantly but I have not tried that I must say. You can try it out rather easily.

Just my 5 cents....

ramiss, 2020-06-11:
Excellent thoughts Peter and point well taken about using both for different reasons.  Thanks again and thanks for this repository, it has certainly been a life saver.

jps2000, 2020-06-11:
welcome

closed 2020-06-11 by jps2000