jps2000/BNO080 #13: Receiving data from a sensor at a given frequency

brightproject, 2025-01-01:
Good day @jps2000 
Thank you for your work🙂
I have a microcontroller `stm32f411ceu6` and a sensor `bno08x`
I do assembly and writing of code in `vscode` + `platformio`.
The code is:
```
#include "Arduino.h"
#include <Wire.h> 

#define SERIAL_PRINT

#define stm32f411

#define  SERIAL_PORT_SPEED  230400
#define  I2C_SPEED_SET  (400000L) // kHz

#ifdef stm32f411
#define btn_TARE                  PA0              // input pin for button TARE
#endif

int plot_interval                 = 2;         // plot interval in ms
unsigned long plot_timer          = 0;

#define BNO_ADDRESS               0x4B

#define CARGO_SIZE 32
uint8_t cargo[CARGO_SIZE];                                // need to be big enough !

float Q0,Q1,Q2,Q3,H_est;                                     
uint8_t stat_;  

const uint8_t quat_report        = 0x05;          // defines kind of rotation vector (0x05), geomagnetic (0x09), AR/VR (0x28)
                                                  // without magnetometer : game rotation vector (0x08), AR/VR Game (0x29)
const int reporting_frequency    = 400;           // reporting frequency in Hz  // note that serial output strongly reduces data rate

uint32_t rate              = 1000000 / reporting_frequency; 
uint8_t B0_rate            = rate & 0xFF;                       //calculates LSB (byte 0)
uint8_t B1_rate            = (rate >> 8) & 0xFF; 
uint8_t B2_rate            = (rate >> 16) & 0xFF; 
uint8_t B3_rate            = rate >> 24;                        //calculates MSB (byte 3) 

long current_timestamp = millis();
struct euler_t {
  float yaw;
  float pitch;
  float roll;
} ypr_float;

/******* Conversions *************/

#define QP(n)                       (1.0f / (1 << n))                   // 1 << n ==  2^-n
#define radtodeg                    (180.0f / PI)

// function declaration
void quaternionToEuler(float qr, float qi, float qj, float qk, euler_t* ypr, bool degrees);
void get_QUAT();
void set_feature_cmd_QUAT();
void TARE();
void ME_cal(uint8_t P0, uint8_t P1, uint8_t P2, uint8_t P4);
void save_periodic_DCD();
void get_Product_ID();
int yawToHeading(int yaw);
void telemetry_bno08x_min();

void setup() {
  pinMode(btn_TARE, INPUT);
// communication 
  Serial.begin(SERIAL_PORT_SPEED); 
 
  #ifdef stm32f411
[... 314 lines cut ...]
If you rotate the sensor in roll or pitch quite sharply, then gaps in the Euler angles are noticeable, which I calculate using a function from quaternions.
The gaps are quite large, from `5` to `15` degrees depending on the angular velocity of rotation.
If you sample plot_interval every 10 ms, then gaps in angles still occur, although how can you get an angle of, say, `5 degrees` from the sensor in `100 Hz` at first and `15 degrees` after `10 ms` ... there is clearly something with the data output from the sensor.
And the question is probably not about your code, but about the sensor as a whole.
The `I2C` bus speed is `400 kHz`, checked with a signal analyzer.
Maybe you can tell me a way to reliably receive data from the sensor at a given frequency?
I plan to record the obtained data on a `SD card`, but with orientation angles missing by several degrees, this is no good.

Please tell me what the parameter gives

> const int reporting_frequency = 400;

I changed it to `50` and `1500`, but it seems there are no changes in the code.

And the last question, could you tell me how to set the initial position of the quaternion, my board with the sensor has an inverted installation, and when I run the code I get an inverted roll angle.

[image]

I use the `TARE()` function, but I want the code to initially give approximately the correct orientation when powering the microcontroller.
The `Adafruit` library code seems to have such a functionality, something like this:

> sh2_Quaternion_t sh_orientation = { 1, 0, 0, 0};

Tell me, do you know where in your code I can set the default orientation of the quaternion?
I will be glad to chat with you for smart discussions.

jps2000, 2025-01-02:
Hi,
Long time since I worked on that. 
What I remember is, that the communication  between BNO and host is most frequently the problem. The issue is that if the BNO can not send all the data for versatile reasons ( timing budget, hardware bus problems etc.), it restarts sending  all data again.  Best is to look with a scope on the Interrupt HINT_N.  See spec. The line goes L if the BNO has data to report. If this line is H then all data has been transferred. Max data rate is 400Hz and this needs to be configured in the setup.  That means also the frequency on the HINTN line  will be the same. So in case you get less something is wrong!
Ypur results suggest that when you say 50Hz does not make a difference. 1500Hz is out of range anyhow ( 400Hz is max)
This can be  a bus problem (Wires too long, too much capacity, Pullups too low, Clock stretching not handled correctly etc)
Some recommend e.g. 4k7 pullup on clock and 2k2 on Data. Reason is to have SDA set before SCL goes H.
I tried I2C  clock of 1MHz worked fine although just 400kHz is specified.
So  set bus speed to 400khz minimum and reporting frequency to 400Hz and watch  the HINTN line.

brightproject, 2025-01-02:
> Long time since I worked on that.
> What I remember is, that the communication between BNO and host is most frequently the problem.

Thank you for the clarification, I will think about using `INT` interrupts.
Could you tell me how to set the sensor position at startup?
In my device, the board with the sensor is turned `180 degrees`.
Now I can use the function:
```
void TARE(){
uint8_t RV; // rotation vector used to tare defined in quat_report
if(quat_report == 0x05) RV = 0x00;
if(quat_report == 0x08) RV = 0x01;
if(quat_report == 0x09) RV = 0x02;
uint8_t tare_now[16] = {16,0,2,0,0xF2,0,0x03,0,0x07,RV,0,0,0,0,0,0}; //0x07 means all axes 0x04 = Z axis only; based on rotation vector
 uint8_t tare_persist[16] = {16,0,2,0,0xF2,0,0x03,0x01,RV,0,0,0,0,0,0,0};
 I2c.write(BNO_ADDRESS, sizeof(tare_now)+1, tare_now, sizeof(tare_now));

 //I2c.write(BNO_ADDRESS, sizeof(tare_persist)+1, tare_persist, sizeof(tare_persist)); // uncomment for tare persist;
 // NOTE: to undo tare persist requires calibration of the device followed by another tare persist with the device oriented y = north bfz = vertical before.

 }
```
When activated by pressing the button, the bank angle changes from `-180` to `0` degrees, but I need to press the button only when the sensor is oriented in a certain direction, otherwise the rolls and pitches will be incorrect, because the quarnion is determined incorrectly.
It's quite difficult to explain in words.
Also, regarding the function above, I couldn't get it to save data in memory using tare_persist.
I found some information on axis reorientation, but I also haven't been able to figure it out yet
```
// #define Q_FLIP	0.70710678118654752440084436210485
/* BNO_TareSetReorientation(Q_FLIP, 0.0, 0.0, -Q_FLIP)
// This command instructs SH-2 to set the current run-time sensor reorientation.
// The rotation vector is a signed, 16-bit 2’s-complement fixed point number with a Q-point of 14.
HAL_StatusTypeDef BNO_TareSetReorientation(const double X, const double Y, const double Z, const double W) {
 resetHeader(REPORT_COMMAND_REQUEST);
 bufferIO[5] = cmdSeqNo++;
 bufferIO[6] = COMMAND_TARE; // 0x03 – Tare command
 bufferIO[7] = COMMAND_TARE_REORIENT; // 0x02 – Set Reorientation
 *(int16_t *)&bufferIO[8] = (int16_t)(X * SCALE_TO_Q14);
 *(int16_t *)&bufferIO[10] = (int16_t)(Y * SCALE_TO_Q14);
 *(int16_t *)&bufferIO[12] = (int16_t)(Z * SCALE_TO_Q14);
 *(int16_t *)&bufferIO[14] = (int16_t)(W * SCALE_TO_Q14);
 return sendPacket(CHANNEL_CONTROL);
} */

// X Y Z Qw Qx Qy Qz
// South West Down 0 -(√2)/2 (√2)/2 0

 // uint8_t tare_set_reorient[16] = {16,0,2,0,0xF2,0,0x03,0x02,Q_FLIP,0,0,0,0,0,-Q_FLIP,0};
 // int8_t tare_set_reorient[16] = {16,0,2,0,0xF2,0,0x03,0x02,11585,0,0,0,0,0,-11585,0};
```

Information from

> https://www.grozeaion.com/electronics/stm32/stm32-i2c-library-for-bno08x-9-axis-imu

Perhaps you remember how you reoriented the sensor placement in your project🙂?

jps2000, 2025-01-03:
You do not necessarily use INT in the code. Use it for diagnosis first.
To change the orientation you can use just quaternion multiplication as shown in the utilities  file Line 178.
You need to multiply the measured quaternions just with the mapping quats from the spec before calculating the Eulers.

brightproject, 2025-01-03:
> To change the orientation you can use just quaternion multiplication as shown in the utilities file Line 178.

I was hoping to see the code, since you understand your program better than anyone else.
Unfortunately, I don't understand the sensor manual very well and only use third-party libraries to work with `bno08x`.
Please tell me more about the code in line `178`:
https://github.com/jps2000/BNO080/blob/d91fb8be03ed06f16d5394584128aaf37fa57260/utilities#L178

I don't understand yet how to use this code ...

Now I do it like this:

> get_QUAT();

and substitute it into my formula:

> quaternionToEuler(Q0, Q1, Q2, Q3, &ypr_float, true);

```
struct euler_t {
  float yaw;
  float pitch;
  float roll;
} ypr_float;

void quaternionToEuler(float qr, float qi, float qj, float qk, euler_t* ypr, bool degrees = false) {

    float sqr = sq(qr);
    float sqi = sq(qi);
    float sqj = sq(qj);
    float sqk = sq(qk);

    ypr->yaw = atan2(2.0 * (qi * qj + qk * qr), (sqi - sqj - sqk + sqr));
    ypr->pitch = asin(-2.0 * (qi * qk - qj * qr) / (sqi + sqj + sqk + sqr));
    ypr->roll = atan2(2.0 * (qj * qk + qi * qr), (-sqi - sqj + sqk + sqr));

    if (degrees) {
      ypr->yaw *= RAD_TO_DEG;
      ypr->pitch *= -RAD_TO_DEG;
      ypr->roll *= RAD_TO_DEG;
    }
}

Serial.println(ypr_float.roll);
Serial.println(ypr_float.pitch;
Serial.println(ypr_float.yaw);

```

[image]

I have such a sensor, in the device I turn the board over and the `Z axis` looks down.

When I start the device for the first time, the roll angle is `-180` degrees.
Next, I need to find an imaginary north and point the Y axis at it and press `TARE`.
Immediately, the orientation angles become `0 0 0` and I get the correct definition of the board orientation.
If, before pressing `TARE`, I do not point the Y axis at the imaginary north, then the orientation angles will also become `0 0 0` , but as soon as I start tilting the sensor, the orientation angles are determined incorrectly, when tilting along the roll, the pitch angle will change and vice versa.
When I used the code `adafruit_bno08x`, before initializing the sensor, I specified the orientation of the quaternion and the sensor immediately started and gave the correct orientation and showed the correct `Euler` angles when rotating.
It is still difficult for me to figure out the sensor, it is cool, but quite difficult to learn.
Previously I worked with the `BNO055` sensor and it had a very simple interface and working with it did not cause any problems.
Thank you for your help🙂

jps2000, 2025-01-04:
What I meant before is:
With calling getquat you get your quaternions.
Then insert the following piece of code 
// define the mapping quaternions from the specification // South West Down 0 -(√2)/2 (√2)/2 0 
t0 = 0.0; t1=-0.707 ;  t2= 0.707; t3 =  0.0; 
  qt0 = q0 * t0 - q1 * t1 - q2 * t2 - q3 * t3;      // multiplication matrix
  qt1 = q0 * t1 + q1 * t0 + q2 * t3 - q3 * t2;
  qt2 = q0 * t2 - q1 * t3 + q2 * t0 + q3 * t1;
  qt3 = q0 * t3 + q1 * t2 - q2 * t1 + q3 * t0;
// then use these for calculating Eulers

So re- orientation of the sensor is done always  by applying a quat multiplication with  those mapping quats.
Tare alone is NOT doing a re-orientation of the sensor. It just sets the quats to 1,0,0,0 in any position without having a certain axis of the sensor pointing to north for example

brightproject, 2025-01-09:
> So re- orientation of the sensor is done always by applying a quat multiplication with those mapping quats

Good day🙂
1. I'm trying to rotate the sensor axes to get the correct orientation angles from the quaternion.

I'm using this code
```
float t0,t1,t2,t3;
float Q0,Q1,Q2,Q3;
float QT0,QT1,QT2,QT3;

t0 = 0.0; t1=0.707 ; t2= 0.707; t3 = 0.0;
// t0 = 0.707; t1= 0.0 ; t2= 00; t3 = 0.707;
QT0 = Q0 * t0 - Q1 * t1 - Q2 * t2 - Q3 * t3; // multiplication matrix
QT1 = Q0 * t1 + Q1 * t0 + Q2 * t3 - Q3 * t2;
QT2 = Q0 * t2 - Q1 * t3 + Q2 * t0 + Q3 * t1;
QT3 = Q0 * t3 + Q1 * t2 - Q2 * t1 + Q3 * t0;
// then use these for calculating Eulers

// passing the quaternion AFTER the multiplication matrix - inverted sensor orientation
quaternionToEuler(QT0, QT1, QT2, QT3, &ypr_float, true);

void quaternionToEuler(float qr, float qi, float qj, float qk, euler_t* ypr, bool degrees = false) {

    float sqr = sq(qr);
    float sqi = sq(qi);
    float sqj = sq(qj);
    float sqk = sq(qk);

    ypr->yaw = atan2(2.0 * (qi * qj + qk * qr), (sqi - sqj - sqk + sqr));
    ypr->pitch = asin(-2.0 * (qi * qk - qj * qr) / (sqi + sqj + sqk + sqr));
    ypr->roll = atan2(2.0 * (qj * qk + qi * qr), (-sqi - sqj + sqk + sqr));

    if (degrees) {
      ypr->yaw *= RAD_TO_DEG;
      ypr->pitch *= -RAD_TO_DEG;
      ypr->roll *= RAD_TO_DEG;
    }
}
```

But the success is only half.
Now when turning on the power, my board inverted with the `Z axis` shows the correct orientation, but as soon as I press the button and apply the `TARE` function, the roll angle flips and I get what was before the axes were flipped.
Can I ask you to tell me some things again?
Perhaps, in addition to reorienting the axes via a quaternion, it is necessary to somehow specify this for the `TARE` function?
I am really a little confused with this `BNO08x` sensor, it really does magic and it seems to have gotten out of my control)))
2. As I wrote above, there is another ambiguity with the `TARE` function.
If I turn on the power of the device, and, for example, turn the device over so that it has the default orientation of the board, then the Euler angles are determined perfectly and everything is fine.
But if I press the button and activate `TARE`, then miracles happen.
If I do not direct the X-axis to the conventional north (it may not be magnetic north, you always need to select a position for calibration), just rotate the device clockwise or counterclockwise around the `Z-axis`, then if I orient the sensor incorrectly and apply `TARE`, then tilting the sensor by the roll or pitch angle I do not get the desired angles, because the sensor when changing, for example, only the roll - begins to change both the roll and pitch, and vice versa.
If I manage to orient the board empirically to the desired point - sometimes this is magnetic north, sometimes an arbitrary direction and apply `TARE`, then I calibrate the sensor by the table level, for example, and then the roll or pitch changes correctly.
Is the TARE function not performing correctly or are there other reasons for this behavior?

Perhaps I do not understand the `TARE` functionality correctly, as far as I understand - this feature sets the orientation angles to zero, after installing the sensor on some object, in order to correct the position.
In my understanding, I must perform two steps:

> a. Reorient the axes
> b. Apply TARE

I.e. when turning on the device, the orientation angles must be +-correct, and `TARE` will only add positioning accuracy, but for me `TARE` spoils everything.

Before asking such questions, I conducted a lot of experiments and went through all the possible board reorientations from the documentation table, but nothing works.

brightproject, 2025-01-09:
Addition - decided not to use the `TARE` function until I learn how to use it.
I just reorient the axes, flash the device and turn it on, achieve `stat_=3` and use the Euler orientation angles obtained from the quaternion

```
float t0,t1,t2,t3;
float Q0,Q1,Q2,Q3;
float QT0,QT1,QT2,QT3;

t0 = 0.0; t1=-0.707 ; t2= 0.707; t3 = 0.0;

QT0 = Q0 * t0 - Q1 * t1 - Q2 * t2 - Q3 * t3; // multiplication matrix
QT1 = Q0 * t1 + Q1 * t0 + Q2 * t3 - Q3 * t2;
QT2 = Q0 * t2 - Q1 * t3 + Q2 * t0 + Q3 * t1;
QT3 = Q0 * t3 + Q1 * t2 - Q2 * t1 + Q3 * t0;

// quaternionToEuler(QT0, QT1, QT2, QT3, &ypr_float, true);
quaternionToEulerJps2000(QT0, QT1, QT2, QT3, &ypr_float, true);

void quaternionToEuler(float qr, float qi, float qj, float qk, euler_t* ypr, bool degrees = false) {

    float sqr = sq(qr);
    float sqi = sq(qi);
    float sqj = sq(qj);
    float sqk = sq(qk);

    ypr->yaw = atan2(2.0 * (qi * qj + qk * qr), (sqi - sqj - sqk + sqr));
    ypr->pitch = asin(-2.0 * (qi * qk - qj * qr) / (sqi + sqj + sqk + sqr));
    ypr->roll = atan2(2.0 * (qj * qk + qi * qr), (-sqi - sqj + sqk + sqr));

    if (degrees) {
      ypr->yaw *= RAD_TO_DEG;
      ypr->pitch *= -RAD_TO_DEG;
      ypr->roll *= RAD_TO_DEG;
    }
}

void quaternionToEulerJps2000(float q0, float q1, float q2, float q3, euler_t* ypr, bool degrees = false) {

// Euler angles ( Tait-Bryan )from quaternions 
 
 ypr->yaw   =  atan2(q1 * q2 + q0 * q3, q0 * q0 + q1 * q1 - 0.5f);   // heading
 ypr->pitch = -asin(2.0f * (q1 * q3 - q0 * q2));
 ypr->roll  =  atan2(q0 * q1 + q2 * q3, q0 * q0 + q3 * q3 - 0.5f);
 

    if (degrees) {
      ypr->yaw *= RAD_TO_DEG;
      ypr->pitch *= -RAD_TO_DEG;
      ypr->roll *= RAD_TO_DEG;
    }
}
```

In the code you can see that I use two functions for calculating `Euler` from a quad, they look different, but they calculate the same - also another mystery of the quaternion😄

jps2000, 2025-01-10:
Hi
Good that you are making progress.
I still do not  know what your intention is to use the tare function.
There ar basically 2 tare options you can tare all axes
and you can tare just the Z  axis ( heading)
this is coded by:  0x07 or 0x04 as mentioned in my code:

uint8_t tare_now[16] = {16,0,2,0,0xF2,0,0x03,0,0x07,RV,0,0,0,0,0,0}; //0x07 means all axes 0x04 = Z axis only; based on rotation vector

There are  in total 12 methods of calculation Eulers. you find all in the utilities. This is  because Euler angles depend on the order doing rotations around the x,y,z axes.

Is the data rate solved now?

brightproject, 2025-01-12:
> Good that you are making progress.

This is all thanks to your work with the [library](https://github.com/jps2000/BNO080)🙂

> I still do not know what your intention is to use the Tare function.

I wanted to compensate for the inaccurate installation of the sensor inside the housing and on the object, but for now I have abandoned this idea.

> this is coded by: 0x07 or 0x04 as mentioned in my code:

I used Tare with `0x07`, if I specify `0x04` then nothing happens at all and does not change when I use `Tare`.

It is interesting to use connection by `SPI` and use `INT` pin, and try it all on `ESP32`.

> https://github.com/myles-parfeniuk/esp32_BNO08x

jps2000, 2025-01-12:
Thank you . Can you please give it a star :-)
Installation should be manageable to be acurrate enough. 
It may be helpful to place the board on a  piece of foam and do levelling with 3 plastic screws. When done secure everything with hot glue.
Advantage is that you get  some damping of noise  that can overload the sensors
Tare with 0x04 should work but  I think you need to have  stat_  =  3 that it does it.

closed 2025-02-08 by jps2000